jalcok1/capstone #1: Block diagram corrections

SethMWard, 2015-01-14:
Below is a list of corrections and refinement that needs to be done to the block diagram document as discussed in the weekly meeting on 1/13/2015.
 
Level 0 corrections
Input 
1. Change USB OTG to High Speed Upstream
2. Remove ADC, DAC, I2C, CAN, and UART from inputs
3. Change SWJ to SWD
4. Add USB Full Speed Debug to inputs

Output 
1. Change USB OTG to USB Full Speed
2. Change Digital out to Digital I/O
3. Remove SWJ

Level 1 corrections
Input corrections
1. Change SWJ to SWD
2. Change USB 2.0 to USB High Speed
3. Change Wall wart value to 5V

Output corrections
1. Add USB Full Speed
2. Add ADC 
3. Remove Ethernet (Ethernet and USB connections are either or)

Level 2 corrections

1. Change SPS/LDO connection from serial to parallel

SethMWard, 2015-01-14:
Assigned to myself but wanted to hear any input you have before starting.

jalcok1, 2015-01-15:
Added corrections to level 0 block diagram and table in FunctionDiagram.docx.  Once level1 and level2 changes have been incorporated we can do review on the document
>

jalcok1, 2015-02-08:
Block diagrams look good for now.  If any changes are made in the future the block diagrams will need to be updated.

closed 2015-02-08 by jalcok1